redpomodoro/byd_battery_box #2: Feature: Add sensor which indicated if battery is in balancing mode

vistalba, 2024-12-24:
Like Olli's "BYD-Logger" add a Sensor if the battery is in balancing state. (At least one cell is currently balanced).

[image]

redpomodoro, 2024-12-27:
Looking at this but it will take some time to figure out a good way to get this data into device structure of home assistant.

vistalba, 2024-12-27:
I think (just for this one) it's a boolean sensor which is just true when balancing is active. 

The question is more, do you get this information already over Modbus RTU over TCP or do we need to do some more reverse engineering, how we can fetch this data?

redpomodoro, 2024-12-30:
Currently I have a new dev version which provides the number of cells that are balancing. 

As reference information I use several other projects that fetch data from the battery. So currently no additional reverse engineering is needed, but these sources conflict sometimes so it takes time to figure out what is accurate. 

Hope to release a new version tomorrow.

vistalba, 2024-12-31:
I found also plenty of different implementations and tried to get it how it works. 
But as I understood the code, most of them used other register to get this data. So I was confused.
That's why I started with the reverse engineering of Olli's tool (as it is the one I was most familiar with until yet).

Looking forward to the new version, specially for the fixed "charged" and "discharged" energy and the balancing stuff. Thanks a lot.

redpomodoro, 2025-01-03:
released in v0.1.2

redpomodoro, 2025-01-20:
Improved data structure for the individual balancing in similar format as the temperatures and voltages so it is split by module. Will go out in 0.1.5 release.

redpomodoro, 2025-01-22:
Closing as released in v0.1.5

closed 2025-01-22 by redpomodoro