redpomodoro/byd_battery_box #3: Sensors sensor.byd_charge_total_energy and sensor.byd_discharge_total_energy does show unaccurate values

vistalba, 2024-12-26:
[image]

Sensors sensor.byd_charge_total_energy and sensor.byd_discharge_total_energy does show unaccurate values.
In comparison with the BYD Logger App which shows:

[image]

Do we know why this difference exists? 

```

redpomodoro, 2024-12-27:
Issue was caused that the values were converted use Big Endian bytes instead of Little Endian bytes. Fix is done in dev, will be going out in next release.

vistalba, 2024-12-27:
You're great! Possible to buy/donate you somewhere a coffee?

redpomodoro, 2024-12-30:
Thanks for the coffee! You already found the link before I could post it here. Really appreciate it!

redpomodoro, 2024-12-31:
Released

vistalba, 2025-01-01:
[image]

Updated. Most of the Sensors in BMS1 section show probably wrong values. (temperatures are wrong, voltage max/min are vice versa?, efficiency, SoH, SoC and energy charged/discharged)
Does this work for your battery? Which type do you have?

redpomodoro, 2025-01-03:
The wrong numbers are probably caused by using multiple applications connecting to the battery simultaneously. The new version 0.1.3 has better connection handling and checks for when this should occur, but it is hard to totally avoid. 

You can see can see that it works for me in the screenshot in the readme page. Later I will update apex charts code so that it shows the values for all modules and add a chart for the temperatures.

vistalba, 2025-01-03:
> The wrong numbers are probably caused by using multiple applications connecting to the battery simultaneously. The new version 0.1.3 has better connection handling and checks for when this should [...]

This could be probably as sometimes I also run Olli's tool. So I will try without this and report back if there are still issues in a new specific post :)
Already tried to use a modbusproxy to aviod issues. But I wasn't successful so far.

vistalba, 2025-01-05:
Works as expected with just a single modbus connection. Thx.

closed 2025-01-05 by vistalba