redpomodoro/byd_battery_box #10: Cell Balancing and voltage history

vistalba, 2025-01-25:
As an improvement I would like if there is a cell balance history. Instead of showing numbers 1,2,4,8,16,32,64,128,256,512...32768 I would just use "true" or "false". As the numbers are not really useful, right?

[image]

Why not adding an entity per cell e.g.:

```
sensor.bms_1_module_1_cell_1
sensor.bms_1_module_1_cell_2
sensor.bms_1_module_1_cell_3
sensor.bms_1_module_1_cell_4
sensor.bms_1_module_1_cell_5
sensor.bms_1_module_1_cell_6
sensor.bms_1_module_1_cell_7
sensor.bms_1_module_1_cell_8
sensor.bms_1_module_1_cell_9
sensor.bms_1_module_1_cell_10
sensor.bms_1_module_1_cell_11
sensor.bms_1_module_1_cell_12
sensor.bms_1_module_1_cell_13
sensor.bms_1_module_1_cell_14
sensor.bms_1_module_1_cell_15
sensor.bms_1_module_1_cell_16
sensor.bms_1_module_2_cell_1
sensor.bms_1_module_2_cell_2
...
```
This would also be interesting for cell voltage, so we would be able to track if always the same cell is high/low. This could then be feeded to influx to track single cell voltage history long term.

redpomodoro, 2025-01-27:
There was a bug in the code, it should be returning 1 and 0's ie true or false. Fix will go out in next release with also updated code for displaying the table.

I am unsure if setting up different sensors for each sensor is a good approach as it will be loads of sensors. Especially if you have 3 towers with 8 modules. Maybe there is another way to get stats on cells balancing.

vistalba, 2025-01-27:
Basically, I agree with having too many sensors. For a 8 Module HVM this would led to 128 sensors just for voltage. There is an option to deactivate sensors this by default when installing the integration. May this could be an option?
I’m not exactly sure how I would be able to track all cell voltages in InfluxDB otherwise.

redpomodoro, 2025-01-29:
If you want to track into influx you can setup the influx integration: https://www.home-assistant.io/integrations/influxdb/ and add something like this to the configuration:

entity_globs:
      - sensor.bms_*_cells_balancing
      - sensor.bms_*_cells_average_voltage

This should also store the attributes which in influx you can split up into individual cells and do further processing.

However given that the data provided by the integration is only a snapshot updated every 10 minutes, it could miss cells that have been balanced within that time period. So I'm not sure if you did all this data processing would lead to useful results.

vistalba, 2025-01-29:
I've set to update it every 120 seconds when I installed the integration. But it doesn't seems to work... but I don't know why yet.

redpomodoro, 2025-01-30:
When working on the logs I found that it logs exactly tell which cells are being balanced, this will be a better source in case you want to track history. I will release a new version soon so you can start try experiment with it.

closed 2025-03-11 by redpomodoro