redpomodoro/byd_battery_box #15: Sensors do not update ob v0.1.6

vistalba, 2025-02-10:
Since update today morning to v0.1.6 sensors stuck and do not update at all for multiple hours.
Downgrade to v0.1.5 has solved the issue.
After another update that issue reappears. So for me at least it looks related to v0.1.6.

redpomodoro, 2025-02-11:
Released a few bug fix release. But I guess it was caused by that the binding could not created the a folder for the log files. Should be fixed in v0.1.9

redpomodoro, 2025-02-11:
Correction you need v0.1.10 forgot to update 1 file...

vistalba, 2025-02-11:
Updated to v0.1.10. 
May you could also approve this PR? https://github.com/redpomodoro/byd_battery_box/pull/14
Fixes the mV / V issue with avg cell voltage.

vistalba, 2025-02-11:
Works now for me too.

vistalba, 2025-02-11:
I tried different things... also reinstall the integration from scratch. As soon as v0.1.10 is used (instead of v0.1.5) the sensor updates and after a short period of time it stucks on a value forever.
Stats also showing this:

[image]

At the begin the "flat line" at exactly the same voltage over multiple hours. 
At the right, there is some missing measurements at all. 
Downgrading to 0.1.5 solves the problem. No other modbus client is running.
Any hint how to troubleshoot?

vistalba, 2025-02-11:
There are multiple.... but not really useful in my opinion.. it just indicated that it is not connected.

```
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: pymodbus.logging
Quelle: custom_components/byd_battery_box/extmodbusclient.py:43
Integration: BYD Battery Box (Dokumentation, Probleme)
Erstmals aufgetreten: 14:35:15 (1 Vorkommnisse)
Zuletzt protokolliert: 14:35:15

Failed to connect
```
```
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.byd_battery_box.extmodbusclient
Quelle: custom_components/byd_battery_box/extmodbusclient.py:49
Integration: BYD Battery Box (Dokumentation, Probleme)
Erstmals aufgetreten: 14:35:15 (1 Vorkommnisse)
Zuletzt protokolliert: 14:35:15

not able to connect to 192.168.16.254:8080
```
```
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.byd_battery_box.bydboxclient
Quelle: custom_components/byd_battery_box/bydboxclient.py:76
Integration: BYD Battery Box (Dokumentation, Probleme)
Erstmals aufgetreten: 14:35:15 (1 Vorkommnisse)
Zuletzt protokolliert: 14:35:15

Error reading base info 192.168.16.254:8080 unit id: 1
Traceback (most recent call last):
  File "/config/custom_components/byd_battery_box/bydboxclient.py", line 76, in init_data
    result = await self.update_info_data()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/byd_battery_box/bydboxclient.py", line 253, in update_info_data
    regs = await self.get_registers(address=0x0000, count=20)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/byd_battery_box/extmodbusclient.py", line 79, in get_registers
    data = await self.read_holding_registers( unit_id=self._unit_id, address=address, count=count)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/byd_battery_box/extmodbusclient.py", line 74, in read_holding_registers
    return await self._client.read_holding_registers(
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        address=address, count=count, slave=unit_id
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/mixin.py", line 107, in read_holding_registers
    return self.execute(
           ~~~~~~~~~~~~^
        pdu_reg_read.ReadHoldingRegistersRequest(address, count, slave, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/base.py", line 153, in execute
    raise ConnectionException(f"Not connected[{self!s}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.16.254:8080]
```

```
Logger: homeassistant.config_entries
Quelle: config_entries.py:637
Erstmals aufgetreten: 14:35:15 (1 Vorkommnisse)
Zuletzt protokolliert: 14:35:15

Error setting up entry BYD Battery Box for byd_battery_box
Traceback (most recent call last):
  File "/config/custom_components/byd_battery_box/bydboxclient.py", line 76, in init_data
    result = await self.update_info_data()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/byd_battery_box/bydboxclient.py", line 253, in update_info_data
    regs = await self.get_registers(address=0x0000, count=20)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/byd_battery_box/extmodbusclient.py", line 79, in get_registers
    data = await self.read_holding_registers( unit_id=self._unit_id, address=address, count=count)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/byd_battery_box/extmodbusclient.py", line 74, in read_holding_registers
    return await self._client.read_holding_registers(
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        address=address, count=count, slave=unit_id
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/mixin.py", line 107, in read_holding_registers
    return self.execute(
           ~~~~~~~~~~~~^
        pdu_reg_read.ReadHoldingRegistersRequest(address, count, slave, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/base.py", line 153, in execute
    raise ConnectionException(f"Not connected[{self!s}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.16.254:8080]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 637, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/byd_battery_box/__init__.py", line 43, in async_setup_entry
    await entry.runtime_data.init_data(read_status_data = True)
  File "/config/custom_components/byd_battery_box/hub.py", line 93, in init_data
    result = await self._bydclient.init_data(close = close, read_status_data=read_status_data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/byd_battery_box/bydboxclient.py", line 83, in init_data
    raise Exception(f"Error reading base info unit id: {self._unit_id}")
Exception: Error reading base info unit id: 1
```

redpomodoro, 2025-02-11:
Thanks! I found a potential issue when it is reconnecting in case the connection has been dropped. However I'm not 100% this is causing the issue you are experiencing.

I'm running updated code now to see if it is stable and no other issues pop. If ok I will push a new release tomorrow morning.

redpomodoro, 2025-02-12:
Pushed version 0.1.11 out hope this gives more insight.

Thanks for your offer to test a beta version, but I think it is good to make sure the production version is actually working fine.

vistalba, 2025-02-12:
Just updated but it doesn't work for me. Same effect as with v0.1.10.
No update e.g. for `sensor.bmu_power` for already 6 minutes.
After HA restart it get the correct value and since the first value it doesn't change anymore.

`log` folder was created but no file is in this folder currently. So no Log info to share.

```
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.byd_battery_box.bydboxclient
Quelle: custom_components/byd_battery_box/bydboxclient.py:113
Integration: BYD Battery Box (Dokumentation, Probleme)
Erstmals aufgetreten: 08:28:00 (1 Vorkommnisse)
Zuletzt protokolliert: 08:28:00

log did not exist, created new log folder: ./custom_components/byd_battery_box/log/
```
```
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: pymodbus.logging
Quelle: custom_components/byd_battery_box/extmodbusclient.py:48
Integration: BYD Battery Box (Dokumentation, Probleme)
Erstmals aufgetreten: 08:28:09 (1 Vorkommnisse)
Zuletzt protokolliert: 08:28:09

Failed to connect
```
```
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.byd_battery_box.extmodbusclient
Quelle: custom_components/byd_battery_box/extmodbusclient.py:46
Integration: BYD Battery Box (Dokumentation, Probleme)
Erstmals aufgetreten: 08:28:09 (1 Vorkommnisse)
Zuletzt protokolliert: 08:28:09

Retry attempt: 1 connecting to: 192.168.16.254:8080 unit id: 1
```
```
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.byd_battery_box.bydboxclient
Quelle: custom_components/byd_battery_box/bydboxclient.py:165
Integration: BYD Battery Box (Dokumentation, Probleme)
Erstmals aufgetreten: 08:28:47 (1 Vorkommnisse)
Zuletzt protokolliert: 08:28:47

Error reading log data 0
Traceback (most recent call last):
  File "/config/custom_components/byd_battery_box/bydboxclient.py", line 165, in update_all_log_data
    result = await self.update_log_data(unit_id, log_depth=log_depth)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/byd_battery_box/bydboxclient.py", line 545, in update_log_data
    await self._read_log_data_unit(unit_id)
  File "/config/custom_components/byd_battery_box/bydboxclient.py", line 574, in _read_log_data_unit
    new_regs = await self.get_registers(address=0x05A8, count=65)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/byd_battery_box/extmodbusclient.py", line 84, in get_registers
    data = await self.read_holding_registers( unit_id=self._unit_id, address=address, count=count)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/byd_battery_box/extmodbusclient.py", line 79, in read_holding_registers
    return await self._client.read_holding_registers(
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        address=address, count=count, slave=unit_id
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/mixin.py", line 107, in read_holding_registers
    return self.execute(
           ~~~~~~~~~~~~^
        pdu_reg_read.ReadHoldingRegistersRequest(address, count, slave, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/base.py", line 153, in execute
    raise ConnectionException(f"Not connected[{self!s}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.16.254:8080]
```
```
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: pymodbus.logging
Quelle: custom_components/byd_battery_box/extmodbusclient.py:79
Integration: BYD Battery Box (Dokumentation, Probleme)
Erstmals aufgetreten: 08:29:04 (1 Vorkommnisse)
Zuletzt protokolliert: 08:29:04

Cancel send, because not connected!
```
```
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.byd_battery_box.hub
Quelle: custom_components/byd_battery_box/hub.py:130
Integration: BYD Battery Box (Dokumentation, Probleme)
Erstmals aufgetreten: 08:29:04 (1 Vorkommnisse)
Zuletzt protokolliert: 08:29:04

Error reading status data
```

redpomodoro, 2025-02-12:
Thanks! We are slowly getting to the bottom of the issue. The connection gets dropped while loading the log history. 

I'm making some changes that the log history won't be loaded the first time but only the latest entries. To get more entries I'm adding a button to trigger that manually.

Also putting in additional checks in case the connection gets dropped so it can reconnect. 

Testing it overnight and if it seems to work I'll push out another version tomorrow.

redpomodoro, 2025-02-14:
It took more time to get it stable, but new version 0.1.12 is out. Now you have buttons the load the log history and by default it will only retrieve the last 20 logs. If you trigger loading history it will show progress in the log files.

vistalba, 2025-02-14:
v0.1.12 also doesn't work. Initial values are there but already 8 minutes without update. With 0.1.6 update every 15 seconds is working.

Beside the other errors I posted before this one now also pops up. Don't know if this is related.

```
Logger: homeassistant.core
Quelle: core.py:1577
Erstmals aufgetreten: 10:53:32 (2 Vorkommnisse)
Zuletzt protokolliert: 10:53:32

Error running job: <Job onetime listen homeassistant_start <function _async_at_core_state.<locals>._matched_event at 0x7fe1bc763ec0> HassJobType.Callback <_OneTimeListener homeassistant.helpers.start:<function _async_at_core_state.<locals>._matched_event at 0x7fe1bc763ec0>>>
Error running job: <Job onetime listen homeassistant_start <function _async_at_core_state.<locals>._matched_event at 0x7fe1bc780180> HassJobType.Callback <_OneTimeListener homeassistant.helpers.start:<function _async_at_core_state.<locals>._matched_event at 0x7fe1bc780180>>>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 635, in state
    numerical_value = int(value)
ValueError: invalid literal for int() with base 10: 'unknown'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 638, in state
    numerical_value = float(value)
ValueError: could not convert string to float: 'unknown'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1577, in async_fire_internal
    self._hass.async_run_hass_job(job, event)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 940, in async_run_hass_job
    hassjob.target(*args)
    ~~~~~~~~~~~~~~^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 1438, in __call__
    self.hass.async_run_hass_job(self.listener_job, event)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 940, in async_run_hass_job
    hassjob.target(*args)
    ~~~~~~~~~~~~~~^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/start.py", line 44, in _matched_event
    hass.async_run_hass_job(at_start_job, hass)
    ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 940, in async_run_hass_job
    hassjob.target(*args)
    ~~~~~~~~~~~~~~^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/template/template_entity.py", line 517, in _async_template_startup
    result_info.async_refresh()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 1126, in async_refresh
    self._refresh(None)
    ~~~~~~~~~~~~~^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 1312, in _refresh
    self.hass.async_run_hass_job(self._job, event, updates)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 940, in async_run_hass_job
    hassjob.target(*args)
    ~~~~~~~~~~~~~~^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/template/template_entity.py", line 463, in _handle_results
    self.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1148, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1085, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1029, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 642, in state
    raise ValueError(
    ...<5 lines>...
    ) from err
ValueError: Sensor sensor.total_byd_battery_energy_charged has device class 'energy', state class 'total' unit 'kWh' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: 'unknown' (<class 'str'>)
```

redpomodoro, 2025-02-14:
The new entries don't seem to be related to this integration. 

I've pushed v0.1.13. Just to be sure it gets fully updated it might be best to remove the integration from HA before updating.

Once updated could you turn on 'debug logging' and reload the integration? You should see entries like:

updating BMU status
bmu 213.1V -3.9A -831.0899999999999W

The log might give a clue to see what happens just after the last successful update occurs. 

Also note that when you press any of the buttons these updates will be suspended until the log history has been loaded.

vistalba, 2025-02-14:
Where do I find this debug log?

redpomodoro, 2025-02-15:
You can enable debug logging by going to 'Settings' -> 'Devices & services' -> Select 'Byd Battery Box' integration -> Select 'Enable debug logging'. Also on the same page you can reload the integration by clicking the 3 dots and select reload.

vistalba, 2025-02-15:
I've now a big log file but it contains a lot of other data. May I did something wrong? 
Is there a easy way to clean this up?

redpomodoro, 2025-02-15:
If you could post entries from the start up till just after the point it stops updating the sensors that would be great. That could be a lot so if you attach it as file that is fine. 

Cleaning up is difficult because that might also filter out the clues.

vistalba, 2025-02-17:
Currently trying to reproduce. I removed the integration, manually deleted the folder in custom_components.
Restarted HA, installed v0.1.13. I enabled debug logging and then restarted HA again. 
But... since yesterday it looks like it is updating fine. So let's see. 

May also there is a problem, when setting update interval lower than 30sec/600sec when installing the integration? (I tried to use 15sec/120sec before. Could this be a problem?)

redpomodoro, 2025-02-17:
After running some tests it indeed seems to be related to shorter update intervals. Looks like the asynchronous modbus handles errors differently which causes the updates to stop, but that needs more testing.

redpomodoro, 2025-02-18:
With the new version v0.1.14 it should be possible to use shorter update intervals. Only the logs need a minimum of 2 minutes otherwise those updates give unexpected results.

vistalba, 2025-02-18:
v0.1.14 tested. Seems to work fine so far (first 20min). Will now monitor it further. 
I close this issue as it seems to be solved in v0.1.13 (with default update config) and v0.1.14 with custom update config of 15sec/120sec/600sec.

closed 2025-02-18 by vistalba

redpomodoro, 2025-02-18:
Cool! It took only 8 releases to fix it ;-)

vistalba, 2025-02-18:
No problem. I‘m happy that it works. Wasn‘t able to get useful logs. Anyway, the update of the BMS was not working before but now the configured update interval is respected. So this is now fixed too.