redpomodoro/byd_battery_box #16: Use for BYD Premium LVS (here 8.0)

MMelzig, 2025-02-16:
I have tried the current release 0.1.13 once in my Home Assistant setting.
 - The battery is a BYD Premium LVS (with two modules of 4 kWh).
 - The inverter is a Storedge SE10K-RWS from Solaredge.

The installation out of the box only showed BMU data but no BMS data. Towers was 0, modules 2.

I then modified bydboxclient.py. I hard-coded towers and modules to meaningful values.

Variant 1:
 - towers = 1
 - modules = 2
These values seemed obvious to me. However, this only resulted in data for BMS_1. "sensor.bms_1_cells_average_voltage" contained voltages for 2 modules, but only the first contained meaningful data.

Variant 2:
 - towers = 2
 - modules = 1
resulted in data for BMS_1 and BMS_1. "sensor.bms_1_cells_average_voltage" and "sensor.bms_2_cells_average_voltage" now still contain content for 1 module and correspond to the values visible using BeConnect. The values for the BMU are now Towers = 2 and Module = 1, which sounds inappropriate.

In addition to the hard coded values for towers and modules, I have also made a change to the number of cell voltages and cell temperatures.

By the way, I find it very good that there is only one sensor with corresponding attributes for cell voltage/temperature/balancing, as this avoids an explosion in the number of sensors.

Should I have done something different during the installation and the addon would have worked without patching?

[bydboxclient.txt](https://github.com/user-attachments/files/18813860/bydboxclient.txt)

redpomodoro, 2025-02-16:
The existing implementation doesn't handle Low Voltage systems well and it from your result it also seems that with low voltage systems different types of Low Voltage systems can be identified. I've improved the handling so it should now be better in setting the right properties for your system. These changes will go out in 0.1.14 release.

Note that you might need to remove the integration first before installing the new version for it to work as aspected.

MMelzig, 2025-02-16:
Thank you very much in advance.

The challenge is to take into account the many different variants and you can't have all of them available for testing. I would therefore like to outline the differences that I have understood. If this is already known, just ignore me! And of course I only know what I have in my home, one LVS system with 1 tower and 2 modules. 

For the HV? System is
- a module a defined quantity of battery cells
-- which does not contain its own BMS
-- of which up to 8 modules can be connected in series in a tower

- a tower
-- contains the BMS, across all the installed modules

- The overall system (hub)
-- contains the BMU
-- and can integrate up to 3 towers (BMS)

For the LVS system is
- a module also a defined quantity of battery cells
-- but with its own BMS
-- of which up to 6 modules can be connected in parallel in a tower

- a tower
-- presumably only serves to mechanically connect the modules
-- the BMS are already in the modules

- The overall system (hub)
-- contains the BMU
-- and can contain up to 64 modules connected in parallel
-- e.g. 16 Battery-Box LVS 16.0 (i.e. with 4 modules)
-- It is possible that a picklist with BMS_1 to BMS_64 is then corresponding selection field created in the Diagonis area in BeConnect.

As I said, I'm not sure whether the towers in the LVS system have any structuring function for the software. In any case, the difference for implementation in the add-on does not seem insignificant to me. Just whether 3 or 64 BMS have to be expected makes a difference.

redpomodoro, 2025-02-18:
Indeed only have 1 system to test with :-)

New version going out today. Please let me know if you can use it now without any modifications.

MMelzig, 2025-02-18:
Thank you for the fast first adaptation.

The installations was successful and I have now the following results:

The sensor.towers is 1, sensor.modules is 2, both fits to the hardware topology. 

The representation in software is only ONE bms_1 with the following details for the cell values:
state_attr('sensor.bms_1_cells_average_voltage','cell_voltages'): [{'m': 1, 'v': [3318, 3317, 3317, 3316, 3317, 3317, 3317, 3316, 3317, 3318, 3318, 3317, 3318, 3318, 3317, 3317]}, {'m': 2, 'v': [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]}]
state_attr('sensor.bms_1_cells_average_temperature','cell_temps): [{'m': 1, 't': [26, 26, 26, 26, 26, 26, 26, 27]}, {'m': 2, 't': [27, 27, 32, 0, 0, 0, 0, 0]}]

You see, that the second block of voltage values 'm': 2 is 0, because there are no further batterie cells for the bms.
For the temperature it is not so obviosly, because the first values in the second block could be correct.

In the attached screenshot of BeConnect you can see the available pick-list and the list of cell values.
[image]

From my point of view, a mapping is necessary that transfers the physical topology into the appropriate software objects. When I use other names, I would describe it as follows:

            Hardware topology                   Software objects

HVS     Tower                                        BMS
            Module                                     accu_groups ('m':1, 'm':2 in the data structure)

LVS     Tower                                         ?? may be not relevant
           Module                                      BMS with ONE accu_group
                                                              (always 'm':1 in the data structure)

When I changed "bydboxclient.py" at line 261 again to:

#        towers, modules = self.convert_from_registers_int4(regs[16:17])
        towers = 1 (in the last patch it was 2)
        modules = 1

I get the BMS_1 and BMS_2 with the values for the cells:
state_attr('sensor.bms_1_cells_average_voltage','cell_voltages'): [{'m': 1, 'v': [3307, 3306, 3307, 3306, 3307, 3307, 3306, 3306, 3307, 3307, 3307, 3307, 3308, 3307, 3306, 3307]}]
state_attr('sensor.bms_1_cells_average_temperature','cell_temps): [{'m': 1, 't': [25, 26, 26, 25, 25, 26, 26, 26]}]
state_attr('sensor.bms_2_cells_average_voltage','cell_voltages'): [{'m': 1, 'v': [3308, 3307, 3307, 3307, 3307, 3307, 3306, 3306, 3307, 3307, 3307, 3307, 3308, 3306, 3307, 3307]}]
state_attr('sensor.bms_2_cells_average_temperature','cell_temps): [{'m': 1, 't': [24, 25, 25, 25, 25, 25, 25, 26]}]

Then there are a few more cosmetic issues:

sensor.cells_per_module is 8, should be 16
sensor.temperature_sensors_per_module is 16, should be 8
(but for both I don't see an impact of these values in the software results). 

sensor.application is "Backup" should be "On Grid" (same was in verion 1.13)
sensor.inverter is "SOSEN" was and should be "SolarEdge LV"

MMelzig, 2025-02-18:
Markdown destroyed the table. So a second try:

| Type   |  Hardware topology   |  Software objects
| ------  | ----------------------- | --------------------
|  HVS   |  Tower                        |  BMS
|            |  Module                     |  accu_groups ('m':1, 'm':2 in the data structure)
|            |                                   | 
| LVS     |  Tower                        | ?? may be not relevant
|           |   Module                     |  BMS with ONE accu_group (always 'm':1 in the data structure)

And also the code block
....
#        towers, modules = self.convert_from_registers_int4(regs[16:17])
        towers = 1
        modules = 1
....

MMelzig, 2025-02-19:
Just because I can test it on my system I make some changes at two files. In "bydboxclient.py" you make a mapping from "towers" and "modules" to "self._bms_qty" and "self._modules". I think that is the pace there a mapping from pyhsical topology to software elements occure. I have adaptes this.

In "sensor.py" the creation / filling of "bms_x" sensorr occures depending of the amout of towers. I changed it, because I don't know how to use  "self._bms_qty", which should be the better choice.

With this modifications it seems to work fine.

[bydboxclient.py.txt](https://github.com/user-attachments/files/18859508/bydboxclient.py.txt)

[sensor.py.txt](https://github.com/user-attachments/files/18859510/sensor.py.txt)

FoxMike1, 2025-02-25:
@MMelzig 
yes, that is working out of the box. I´m using myself a BYD LVS 8.0 and the original code shows only 1 BMS module instead of the two. With your code all devices (BMU, BMS1 and BMS2) show up. 
One issue, which is the same with the other BYD integration on github, is the total amount of charge_total_energy and discharge_total_energy. the decimals are not correct, I see f.e. discharge of 10.548,1 kWh instead of the correct 105,xx kWh, The same entities for each BMS show the correct values. I don´t know, why both integrations produce the same issue at these two entities. I had to make template sensors to correct the values as a go around. Maybe somebody can solve the issue directly in the code in the future. BYD logger "by Olli" shows those values correct. 

Thank you both  for your work here at github. I will do further testing.

FoxMike1, 2025-02-25:
After testing (with code MMelzig): I got no buttons to load old log BMS2 and balancing total stays at 0, while 8 for BMS1 which could even load the old logs.

Markdown card copy/paste from the github examples show only module 1, 2 is missing.

MMelzig, 2025-02-25:
@FoxMike1, please be aware my code parts are only a sketch where a modification in the original code could be done for a mapping of the pysical topology to the software elements. The mapping fits for the amount of BMS elements. But not for other things like the amount of buttons e.g. for loading logs.

With other word my files together with the rest of the original code is NOT A VALID implementation.