redpomodoro/byd_battery_box #17: invalid handler specified ... is all I get

commentsonly, 2025-02-21:
looked wa too promissing ... and since so I failed get it installed.

Followed the instructions, integration does appear, but when clicking upon it all I get is (see subject)

Anything I missed?
In case it does work, would it realise to find 2 HVS 10.2 towers ?

obert99, 2025-02-27:
Unfortunately the same error upon installation on my device. 
Error occurred loading flow for integration byd_battery_box: cannot import name 'ModbusExceptions' from 'pymodbus.pdu' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/\__init\__.py)

commentsonly, 2025-02-27:
bet it got to with something similar currently most modbus implementations fail .... a change in pymodbus .... even the internal modbus integration which ships with home assistant got updated lasdt week

redpomodoro, 2025-03-03:
This seems to be caused indeed by multiple integration sharing pymodbus. You can try removing ALL integrations using pymodbus and modbus configuratio.yaml (for the build in integration into HA), rebooting HA and then reinstalling the integrations and the modbus configuration yaml.

redpomodoro, 2025-03-03:
new version v0.1.15 that should address this issue

commentsonly, 2025-03-03:
Just trying ..... looks way more promissing, a configuration window does open :-)
But apart from "failed to connect" nothing happens, no matter if using the inbuilt 192.168.16.254 or the routed 199.168.1.x IP from the local network.
But perhaps it's also port 8080 causing the problem, no idea yet.
I do get the login page on both ...... hmm. strange. but not if using port 8080
Same for port :80 which I tried aswell. And since it can't left blank I'm lost.

Perhaps way too many devices try to talk to the battery
(node-red running my charge-strategy since the one bYD builtin is purest crap. Resulted in 2 exchanged module within 1 year, but I'm patient EFT could exchange module year by year if they go bust by themselves.

I also get an error reading base info unit id: 1
well I never knew the battery got a unit id. Tried from 0 - 5 no luck.

Perhaps some more advice needed?

obert99, 2025-03-03:
Works perfectly, now! Thanks a lot! Running on my networks DHCP address on port 8080. No static route necessary.
@redpomodoro : you made my day!

redpomodoro, 2025-03-11:
> Perhaps way too many devices try to talk to the battery (node-red running my charge-strategy since the one bYD builtin is purest crap. Resulted in 2 exchanged module within 1 year, but I'm patient EFT could exchange module year by year if they go bust by themselves.
> 

The battery only seems to accept 1 application to be active at the same time. So try disabling any other application that communicates to the battery.

commentsonly, 2025-03-11:
Well there's surely at least 2 things talking to the battery. The inverter and my node-red.
On the other hand if running Be Connect Plus on my personal computer besides the inverter talking to the battery this works aswell, means it's surely possible to communicate with the very same device from multiple clients.
Not to mention that this even works if interacting from the node-red side.

It all depends if a software tries to establish a permanent connection since that results in being an exclusive connection which likely refuses others from being able to talk to the battery.
Had this with HV Logger on my windows machine which allowed me to configure a permanent connection resulting in Be Connect Plus wasn't able to run besides. If configuring it to let say a 10 second interval multiple software runs smoothly side by side.
Any idea if you code connects exclusive and permanent?