redpomodoro/byd_battery_box #18: Installation problem together with add-on solaredge-modbus-multi

MMelzig, 2025-02-28:
In my test system there is a problem with the installation of this add-on version 0.1.14 (without my patches) together with the https://github.com/WillCodeForCats/solaredge-modbus-multi version 3.1.0 in the Home Assistant configuration

If I install the Solaredge add-on first, the BYD-Box add-on can not be installed afterwards. See [home-assistant_Solarede_then_BYD.log](https://github.com/user-attachments/files/19026335/home-assistant_Solarede_then_BYD.log).

If I then deactivate the Solaredge add-on, the BYD-Box add-on can be installed. However, the Solaredge add-on cannot then be reactivated thereafter. See [home-assistant_BYD_then_Solaredge.log](https://github.com/user-attachments/files/19026345/home-assistant_BYD_then_Solaredge.log).

Without the other add-on, each add-on work individually without any problems.

In my live system the add-on version 0.1.14 (with or without my patches) and solaredge-modbus-multi version 3.0.6 in the Home Assistant configuration

- Core 2024.12.5
- Supervisor: 2025.02.4
- Operating System 14.2

the add-ons work together.

Who can solve the problem?

MMelzig, 2025-02-28:
The Home Assistant config of the test system is

WillCodeForCats, 2025-02-28:
This integration needs to be updated for pymodbus 3.8.3.

redpomodoro, 2025-03-03:
Released new version 0.1.15 that should address this.

MMelzig, 2025-03-03:
Thanks,

it is working now in my test system as well as in the live system.