redpomodoro/byd_battery_box #20: Config flow could not be loaded: {"message":"Invalid handler specified"}

h4m0n, 2025-03-02:
Hallo,
I get this error message when I want to start the integration;

I downloaded over HACS, wanted to add the integration and i got this Error message:
Config flow could not be loaded: {"message":"Invalid handler specified"}

what should i do?

Hardware:
inverter: Kostal Plenticore G3 L
Battery: BYD HVM 8 modules (+-22kW)

from Logs:
Logger: homeassistant.config_entries
Source: config_entries.py:3308
First occurred: March 1, 2025 at 7:27:22 PM (4 occurrences)
Last logged: 11:57:57 AM

Error occurred loading flow for integration byd_battery_box: cannot import name 'ModbusExceptions' from 'pymodbus.pdu' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/__init__.py)

Thanks

redpomodoro, 2025-03-03:
You are probably also using the builtin modbus integration. Released new version 0.1.15 should allow them working side-by-side again.

h4m0n, 2025-03-03:
Bingo, it worked like a charm, Thanks a lot!!

closed 2025-03-03 by h4m0n